bluerobotics/BlueOS #2454: bug: When updating from BlueOS 1.1.1 to 1.2.0-beta.8 the OpenVscode extension stop working and refuses to uninstall

ArturoManzoli, 2024-03-14:
### Bug description

After a BlueOS version update from 1.1.1 to 1.2.0-beta.8 the pre-installed OpenVscode extension refuses to open (theme_style.css fetching gets a 404).
Also refuses to uninstall, showing as not installed on the extensions panel (installed tab), but still showing as installed on the left vertical menu.

### Steps to reproduce

* Install and setup OpenVscode extension;
* upgrade BlueOS from 1.1.1 (maybe other versions too) to 1.2.0-beta.8;
* Try to use OpenVscode;
* Try to reinstall extension;

### Additional context

theme_style.css fetching gets a 404 on devTools/Network

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2024-03-14:
MemoryError. Maybe this is our current memory limitation solution? @Williangalvani what do you think?

Williangalvani, 2024-03-20:
> MemoryError. Maybe this is our current memory limitation solution? @Williangalvani what do you think?

I don't understand why kraken is running over the memory limit. it is just supposed to talk to the docker daemon.

Williangalvani, 2024-03-20:
do we want to lift the memory limit for the Kraken service too, meanwhile? so we have a stabler stable?

joaoantoniocardoso, 2024-03-20:
> do we want to lift the memory limit for the Kraken service too, meanwhile? so we have a stabler stable?

Would it be more stable, or would it make the other services to starve?

patrickelectric, 2024-03-22:
@ArturoManzoli check latest 1.2.0-beta.9, it should fix your issue

patrickelectric, 2025-01-16:
this should be not a problem anymore

closed 2025-01-16 by patrickelectric