bluerobotics/BlueOS #2467: DNS problems with IPV6 in BlueOS

JoaoMario109, 2024-03-19:
### Bug description

When you connect to some some router that supports IPV6 and the raspberry acquires IPV6 on some of its interface you will have a IPV6 DNSs set. If you try to delete them, it will say that was successfully deleted but if you try to pring some host as www.google.com from within the raspberry it will still trying to use the IPV6 DNS and will fail. After this, if you reboot the raspberry, even after deleting the IPV6 DNSs, the blueos internet show it again.

### Steps to reproduce

Connect raspberry to some router that supports IPV6 and wait to acquire a IPV6 IP in some of the interfaces.
Check if there are some IPV6 DNSs in the internet icon on BlueOS.
Try to delete some of the IPV6 DNS and pring some host like www.google.com from within the raspberry.
Try to reboot and see the IPV6 DNS appears again

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2024-03-20:
It will try IPV6 because even though it doesn't have an IPV6 DNS, it has an IPV6 address, right?

patrickelectric, 2024-03-20:
@joaoantoniocardoso yes, we saw the interface with both ipv6 and ipv4. But it was weird that the dns solver was still using ipv6.
We tried to delete ipv6 dns configuration in blueos to put ipv4 in higher priority (that was already the case). But the system was still enforcing priority over dns solving in ipv6.

rotu, 2024-09-09:
I think it's a bug that the web frontend and `mavlink2rest` only appear to be listening on IPv4.

e.g. `wget -6 localhost:80` fails
e.g. `wget -6 localhost:6040` fails

Some other services apparently succeed:
e.g. `wget -6 localhost:7777` succeeds (File Browser)
e.g. `wget -6 localhost:9120` succeeds (Pardal)

This is a problem because:
1. It is confusing!
2. Resolution depends on how the domain is entered:
   - e.g. `ping localhost` uses the IPv6 address whereas `ping localhost.` uses the IPv4 address.
   - Node.js address lookup seems to work similarly.
3. Many system utilities *do* use IPv6 preferentially.

The first two lines of my `/etc/hosts` are:
```
127.0.0.1       localhost
::1             localhost ip6-localhost ip6-loopback
```

But `getent` prefers IPv6:
```console
$ getent hosts localhost
::1             localhost ip6-localhost ip6-loopback
$ getent hosts localhost.
127.0.0.1       localhost
```

I think it probably makes sense to support IPv6 rather than try and suppress it everywhere.

closed 2024-09-13 by patrickelectric