bluerobotics/BlueOS #2478: bug: Internet Speed Test shows complete but no data is displayed

JoaoMario109, 2024-03-21:
### Bug description

When running internet speed test, the button is enabled again and the status change to `Done` even without any kind of data being displayed

[image]

Console shows some errors

[image]

### Steps to reproduce

Just executed `Internet Speed Test` normally

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2024-03-21:
I've also seen this happening here.

Williangalvani, 2024-03-21:
can you check if the helper had MemoryError?
IIRC it also manifests there as being unable to create new threads