bluerobotics/BlueOS #2518: bug: Default MAV profile has an unconfigured VeryGenericIndicator

ArturoManzoli, 2024-04-06:
### Bug description

The default layout for MAV profile has an unconfigured VeryGenericIndicator instance. On cockpit this causes the configuration screen to pop up on profile change/reset.
[image]
****

### Steps to reproduce

 On Cockpit:
* Clear the browser cache with CTRL+Shift+R;
* Press the "Reset profiles to default configuration" button;
* Select the standard profile for MAV;
* The veryGenericIndicator config pops out.

### Primary pain point(s)

On cockpit this causes the configuration screen to pop up on profile change/reset.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

ArturoManzoli, 2024-04-07:
I might be wrong, but isn't the original profiles object used on the cockpit fetched from the BlueOS during a layout reset?
If so, the original profile object is stored on the BlueOS and has to be changed there.

rafaellehmkuhl, 2024-04-08:
@ArturoManzoli they are all in the Cockpit repo. It's on `src/assets/joystick-profiles.ts`.

ArturoManzoli, 2024-04-08:
The issue was indeed misplaced

closed 2024-04-08 by ArturoManzoli