bluerobotics/BlueOS #2572: BlueOS could be more helpful in showing what is bloating the disk

rafaellehmkuhl, 2024-04-29:
I had a situation now where my disk was full. The problem is: what is bloating the disk?
My first guess was that it could be the docker images, but removing the three unused ones got me only 8% of the disk back.

Looking at the system information, it has this:

[image]

We could put more information here, detailing which are the folders that take the most space in the system.

rafaellehmkuhl, 2024-05-02:
More info on that:

[image]

My system has about 3GB of extensions that I do not use anymore and that I have no way to delete besides with the command line.

ES-Alexander, 2024-05-10:
This is a slightly broader version of #1482. There's some relevant discussion in that issue about how to go about presenting this, and what we'd want it to cover :-)

rafaellehmkuhl, 2024-05-10:
Yup. In my case thought (and probably for most people) the images use a lot of space, but they are not the biggest problem.

I suspect there's something on the Docker overlay system being left.

ES-Alexander, 2024-05-20:
Quoting from [my comment in the other issue](https://github.com/bluerobotics/BlueOS/issues/1482#issuecomment-1911485627):

> ... it would be preferable if we could also display things like the space taken up by old (still installed) versions, and by data / logs stored in the filesystem (per https://github.com/bluerobotics/BlueOS/issues/1640).
>
>Something like one of [these](https://alternativeto.net/software/baobab/) could be useful, although given we want a webUI we might want something like [diskover](https://github.com/diskoverdata/diskover-community). Alternatively we could roll our own with the specific categories and groupings we're interested in - not sure how involved that would be.

rafaellehmkuhl, 2024-05-21:
Found the culprits:

[image]

ES-Alexander, 2024-11-21:
It would be nice if the visualisation also allowed directly managing things that can be deleted (e.g. see logs are taking up lots of space, and click on that slice/box and delete it to delete all the logs).

closed 2025-12-12 by patrickelectric