bluerobotics/BlueOS #2652: bug: boot loop right after setup

herwig-, 2024-06-01:
### Bug description

"Raspi goes into boot loop after installing"

### Steps to reproduce

* Hardware: Raspi-Pi 4 model B,  4GB
* install pi-light os (tested with 64 bit) (pi-imager: with pre-defined wifi, username, pass, ssh enabled, ...)
* update os (apt install update, upgrade)
* install blueos as described here: `https://github.com/bluerobotics/BlueOS/tree/master/install`

result: rspi will boot again and again ... the bootstrap container seams to continue working throughout the reboots ...

### Primary pain point(s)

continuous reboot means the system is not useable at all!

### Additional context

output from `docker ps`
`CONTAINER ID   IMAGE                                  COMMAND                  CREATED          STATUS          PORTS     NAMES
4ef4b2f590f3   bluerobotics/blueos-core:factory       "/bin/sh -c '/bin/ba…"   31 seconds ago   Up 29 seconds             blueos-core
5bb736501fcd   bluerobotics/blueos-bootstrap:master   "/bin/sh -c /main.py"    14 minutes ago   Up 39 seconds             blueos-bootstrap`

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

Williangalvani, 2024-06-03:
Hi! we're actively working on support for bookworm and improving the installation setup. the boot loop should be gone with the latest master. see #2639

patrickelectric, 2024-06-03:
@herwig- that was a temporary problem that we had in master version, try to install a new image or do a docker pull to get the latest version of BlueOS master.

herwig-, 2024-06-08:
Tested it today and it worked. Thx

closed 2024-06-08 by herwig-