bluerobotics/BlueOS #2673: bug: BACKEND_OFFLINE

herwig-, 2024-06-08:
### Bug description

[image]

BlueOS Version: heads/master-0-g32243cc2 Build: 6/7/2024, 2:51:13 PM

please note, also internet connection failes...

[image]

is the problem maybe related to the image above? In this image it seams that the repository blueos-core is used instead of the BlueOS ?

### Steps to reproduce

Hardware: rpi3, 4gb model B+
OS: light 64bit

sudo bash
apt update

sudo su -c 'curl -fsSL https://raw.githubusercontent.com/bluerobotics/blueos-docker/master/install/install.sh | bash'

rebooot -> then the bootstrap docker Image keeos running but the core:factory docker image always reboots

### Primary pain point(s)

The software does not start as planned

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2024-06-11:
Hi, 64-bit Arm Linux support is still underway. Can you use the old 32-bit version?

> is the problem maybe related to the image above? In this image it seams that the repository blueos-core is used instead of the BlueOS ?

`blueos-core` is correct.

herwig-, 2024-06-12:
thx, will try the 32-bit version asap

herwig-, 2024-06-15:
thx, the 32 bit Legacy version worked (after some tries) -> could anybody add a note on the current state of adaption for the available RPi OS versions?
I know that it is impossible to test all systems but it would be good for a newcommer to see which configuration is activley tested and will likely be the best place to start ...

closed 2024-06-15 by herwig-