bluerobotics/BlueOS #2688: MAVP2P can go crazy (and make everything not work)

rafaellehmkuhl, 2024-06-14:
The following is a capture from Tony's ROV terminal, in the Autopilot panel:

https://github.com/bluerobotics/BlueOS/assets/6551040/ef568cc6-891c-4e62-b237-dd369dfff87b

As can be seen, MAVP2P is going crazy and restarting channels like there was no tomorrow. Result: the ROV could not receive any commands from the mavlink2rest interface. Because of that, Cockpit was not working.

It was possible to command through QGC, which indicates that some of the endpoints were still working fine.

I'm marking that as a critical bug.

Williangalvani, 2024-06-25:
Sounds like we had a regression at some point, as MAVP2P was a valid option to avoid the "MYGCS 255: Heartbeat Lost" issue

rafaellehmkuhl, 2024-06-25:
> Sounds like we had a regression at some point, as MAVP2P was a valid option to avoid the "MYGCS 255: Heartbeat Lost" issue

Tony said he prefers using MAVP2P exactly because of that message not being there all the time, but right now it is breaking the usage of Cockpit.

patrickelectric, 2025-01-09:
@rafaellehmkuhl is this still a valid problem ?

rafaellehmkuhl, 2025-01-12:
> @rafaellehmkuhl is this still a valid problem ?

I don't know. Haven't used anything besides mavlink-server since its introduction on July.

patrickelectric, 2026-02-11:
Moved to mavlink-server