bluerobotics/BlueOS #2698: Wizard didn't automatically start

joaoantoniocardoso, 2024-06-18:
Today I flashed a new BlueOS, and entered our 192.168.2.2, but to my surprise, no wizard dialogue showed up. I waited for more than five minutes, changed pages, and it only started after I refreshed (hit F5)

BlueOS Version: tags/1.3.0-beta.5-0-geb70c041 
Browser: 127.0 (64-bit)

patrickelectric, 2024-06-18:
> Today I flashed a new BlueOS, and entered our 192.168.2.2, but to my surprise, no wizard dialogue showed up. I waited for more than five minutes, changed pages, and it only started after I refreshed (hit F5)
> 
> BlueOS Version: tags/1.3.0-beta.5-0-geb70c041 Browser: 127.0 (64-bit)

The wizard level is stored in the frontend, not in the vehicle, this should be changed

JoaoMario109, 2024-06-19:
@joaoantoniocardoso I could not reproduce this here, do you know if the wizard entry was in the bag?

patrickelectric, 2024-06-19:
After testing here, it appears that wizard is already storing in the bag, unable to replicate.

joaoantoniocardoso, 2024-06-19:
My computer was heavily loaded, possibly invalid indeed.

closed 2024-06-19 by joaoantoniocardoso