bluerobotics/BlueOS #2700: Wizard should constantly check internet connection

joaoantoniocardoso, 2024-06-18:
There's the possibility of losing the internet after the internet step, so we probably should go back to that step if the connection fails.

_Originally posted by @joaoantoniocardoso in https://github.com/bluerobotics/BlueOS/issues/1830#issuecomment-1611394693_
            

https://user-images.githubusercontent.com/5920286/340807954-6ff1a1e8-990b-4e31-bc69-ee0a61b304e3.mp4

 BlueOS Version: tags/1.3.0-beta.5-0-geb70c041

patrickelectric, 2024-06-19:
@joaoantoniocardoso the video does not exist.

patrickelectric, 2024-06-19:
@JoaoMario109 can you confirm that this is still an issue ? As you said, this appears to be fixed already in master (#2584 ?)

joaoantoniocardoso, 2024-06-19:
> @joaoantoniocardoso the video does not exist.

fixed

patrickelectric, 2024-07-05:
@joaoantoniocardoso fix is already fixed in master, right ?

joaoantoniocardoso, 2024-07-05:
Still the same problem:

https://github.com/bluerobotics/BlueOS/assets/5920286/fcfa7a1f-6e2b-4c5b-a113-410670791563

patrickelectric, 2024-07-05:
@JoaoMario109 can you take a look ?