bluerobotics/BlueOS #2702: [Wizard] parameters table showing "Unknown" values

joaoantoniocardoso, 2024-06-18:
[image]

patrickelectric, 2024-06-18:
That's really weird!

JoaoMario109, 2024-06-19:
I wasn't able to replicate this, do you have any more info about how to reproduce?

patrickelectric, 2024-06-19:
I saw that once, it appears to be somehow hard to replicate.

joaoantoniocardoso, 2024-06-19:
I have the impression that I got disconnected from the internet just before seeing that

joaoantoniocardoso, 2024-06-19:
update: I just got it again, but only for some parameters. Stable connection this time:

[image]

joaoantoniocardoso, 2024-06-19:
It writes the correct parameters. Are we somehow loading the values to the interface before they are gathered (race condition)?

PR #2783 (2024-07-04): frontend: wizard: DefaultParamLoader: Fix Unknown
Closes #2702
closed 2024-07-04 by patrickelectric