bluerobotics/BlueOS #2704: [Autopilot Paramters] user can scroll down beyond items

joaoantoniocardoso, 2024-06-18:
https://github.com/bluerobotics/BlueOS/assets/5920286/ab26c6f8-dc41-4b26-b6ab-f7fa5599897d

patrickelectric, 2024-06-19:
Unable to replicate with latest master

[image]

joaoantoniocardoso, 2024-06-19:
I just updated and tested again (Crhome Version 126.0.6478.61 (Official Build) (64-bit)):

https://github.com/bluerobotics/BlueOS/assets/5920286/97a04606-bbbf-4966-83aa-109e7340e441

joaoantoniocardoso, 2024-06-19:
My params file:

[Submarine-4.1.2-STABLE-20240521-202146.params.zip](https://github.com/user-attachments/files/15904569/Submarine-4.1.2-STABLE-20240521-202146.params.zip)

joaoantoniocardoso, 2024-06-19:
Can anyone else check if it happens on other systems? @Williangalvani @rafaellehmkuhl @JoaoMario109

patrickelectric, 2024-06-19:
@JoaoMario109 was able to replicate and it's investigating it

closed 2024-06-24 by patrickelectric