bluerobotics/BlueOS #2721: core: frontend: GPS icon is confusing

ES-Alexander, 2024-06-21:
### Current behaviour

There's a crosshair with a number on it, which has no intuitive meaning (e.g. are there 5 locations??)

### Expected or desired behaviour

Use a satellite icon (e.g. [mdi-satellite-variant](https://pictogrammers.com/library/mdi/icon/satellite-variant/)), at least when showing the number of connected satellites.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2024-06-21:
> ### Current behaviour
> There's a crosshair with a number on it, which has no intuitive meaning (e.g. are there 5 locations??)
>
> ### Expected or desired behaviour
> Use a satellite icon (e.g. [...]

This to me is more satellite communication than number of satellites. We can add in the tooltip what the number means. The cross + circle icon is a common GPS position indicator for android.

patrickelectric, 2024-07-05:
This was fixed by #2746

closed 2024-07-05 by patrickelectric