bluerobotics/BlueOS #2743: Bag of holdings is not listed as a service

rafaellehmkuhl, 2024-06-25:
[image]

patrickelectric, 2024-06-25:
Do you have the log of your system ? Which blueos are you running ?

rafaellehmkuhl, 2024-06-25:
Latest master.

Helper logs:

[helper-20240625-171027.gz](https://github.com/user-attachments/files/15975622/helper-20240625-171027.gz)
[logfile_06-25-2024_16:09:37.log](https://github.com/user-attachments/files/15975623/logfile_06-25-2024_16.09.37.log)

patrickelectric, 2024-06-25:
Do you have the log of bag of holding ?

rafaellehmkuhl, 2024-06-25:
> Do you have the log of bag of holding ? It appears to be dead

I can get once I get home, but it is not dead and it log does not contain anything strange.

rafaellehmkuhl, 2024-06-25:
It's probably a race condition.

patrickelectric, 2025-01-11:
Unable to replicate, @rafaellehmkuhl let us know if you are still having this problem

closed 2025-01-11 by patrickelectric