bluerobotics/BlueOS #2747: installer: Only remove blueos-related images

matt-bathyscope, 2024-06-25:
### Current behaviour

I just ran the installer on a Pi that had an older version of BlueOS as well as my own docker container and the installer script blew away all images, not just the BlueOS related images. This is not ideal UX and maybe we should have a warning before doing it.

### Expected or desired behaviour

Either warn before deleting running containers and cached images, or only remove BlueOS-related docker containers and images.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2024-06-26:
right now you can manually remove all BlueOS images and `export NO_CLEAN=1` before calling the installer

patrickelectric, 2024-06-26:
We should add the default value for the variable in the header of the file and also document it in the readme.