bluerobotics/BlueOS #2859: bug: Is there a build guide available on how to get a raspberry pi image?

mindfulme, 2024-08-14:
### Bug description

I can't find any dev guide on how to build a raspberry pi 4 image to flash on sd card

### Steps to reproduce

1. Clone repository
2. Follow readme
3. Fail to build

### Primary pain point(s)

Cant build image 

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2024-08-14:
Hi @mindfulme, did you run the install script ?

mindfulme, 2024-08-14:
no, what to do after I run install script?

patrickelectric, 2024-08-14:
Take a look in https://github.com/bluerobotics/BlueOS#quick-links-, and check the instructions for manual installation.

Is there a reason for not using the provided image ?

mindfulme, 2024-08-14:
Yes, there is. I need to change an address in `Detector.py` file and rebuild the whole code

mindfulme, 2024-08-14:
Chech this thread https://discuss.bluerobotics.com/t/blueos-arduplane-4-5-4-firmware-no-sensors-detected-barometer-not-found/17488/13

patrickelectric, 2024-08-14:
For that, you should be able to modify blueos and build it, after that is done, you should use the version chooser to change the running BlueOS version. 
Check: https://blueos.cloud/docs/blueos/1.1/advanced-usage/#blueos-version

To build you can use buildx with docker or github actions, just a matter to configure your github env variables. 

`DOCKER_USERNAME` and `DOCKER_PASSWORD`.

Vils0n, 2024-08-15:
built with docker completes just fine, only issue is the constant `pull access denied, repository does not exist or may require authorization: server message: insufficient_scope: authorization failed` when `Inspect image` runs

Vils0n, 2024-08-15:
[image]

Vils0n, 2024-08-15:
and build is generated on docker hub, I am able to download it 
[image]

Vils0n, 2024-08-15:
[image]
but where to get `blueos-raspberry.zip` as on screenshot in your official releases

Vils0n, 2024-08-15:
[image]
I guess this is the easiest way to go, to just upload `.tar` file here from docker hub

mindfulme, 2024-08-15:
[image]

@patrickelectric  do you know why it skipped `deploy-raspberry-image` job???

mindfulme, 2024-08-16:
ok, I resolved everything. Had to change the username here in config file
[image]

closed 2024-08-16 by mindfulme

patrickelectric, 2024-08-21:
Hi @mindfulme I'm glad that you figured it out. Sorry for the late reply, I was in a work trip.
About the raspberry image deployment, you need to setup a self-hosted github action in a raspberry pi to have it running.
To install your custom docker images, you just need to go in pirate mode and put your dockerhub repository name: <user>/<docker>