bluerobotics/BlueOS #2871: bug: Unable to fetch firmware or bootloader version on Raspberry Pi B3+

ShootingStars76, 2024-08-26:
### Bug description

When running on Raspberry Pi B3+ when you go to Sys Info > Firmware, no firmware or bootloader version is displayed. 

### Steps to reproduce

1: Turn on Pi 
2: Go to System Infromation 
3: Select firmware 
4: Raspberry pi firmware and bootloader firmware don't display 
5: Throws errors: 
[image]

### Primary pain point(s)

This bug so far does not affect any operations that I know of. However, it does appear to be related to other bugs effecting BlueOs on Rpi B3+ 

### Additional context

[system_logs.zip](https://github.com/user-attachments/files/16754683/system_logs.zip)
Have tried stable/beta branches and multiple sd cards but does not fix the issue.
Uptime does not fix the issue. 

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rotu, 2024-09-11:
Running `sudo rpi-eeprom-update` shows the message "This tool only works with a Raspberry Pi 4". I think this might suggest a contributing cause.

patrickelectric, 2024-09-15:
> Running `sudo rpi-eeprom-update` shows the message "This tool only works with a Raspberry Pi 4". I think this might suggest a contributing cause.

Thanks @rotu!