bluerobotics/BlueOS #2898: question: How do I make a an environment to build image?

mindfulme, 2024-09-15:
### Bug description

`   runs-on: raspbian-armv7-kernel-5.10.33` where do I get it or how to set it up?

` runs-on: blueos-ci`

### Steps to reproduce

Check https://github.com/bluerobotics/BlueOS/blob/d31163bba177495c25ab7d2928418e531d970ed5/.github/workflows/test-and-deploy.yml#L252

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

mindfulme, 2024-09-15:
like this or some other way? https://github.com/wbond/pi-github-runner

patrickelectric, 2024-09-15:
It's just necessary to follow the documentation: https://docs.github.com/en/actions/hosting-your-own-runners/managing-self-hosted-runners/adding-self-hosted-runners

closed 2024-09-15 by patrickelectric