bluerobotics/BlueOS #2921: Internet connection status should be non-binary

Williangalvani, 2024-09-25:
we should have a value for "limited connectivity" where some pings go through and some don't. 
We shouldn't limit functionality unless we are 100% we are offline

ES-Alexander, 2024-10-07:
Marking this as important because not every pinged address is required for every internet-connected service. Services should only be disabled when their relevant domain(s) cannot be connected to[^1], and that information is [already provided in the underlying internet check](https://github.com/bluerobotics/BlueOS/blob/master/core/services/helper/main.py#L482).

It would also be helpful to include a limited connectivity icon (e.g. `mdi-web-minus`) in the status bar, and allow clicking on it to see a summary of which pings domains are failing to respond (and ideally which services cannot be used as a result).

[^1]:As extra context, a ping can be blocked because a site is down, or because the user or their ISP have intentionally blocked it, and us blocking unrelated functionality when that's the case may come across as problematic.

Williangalvani, 2025-02-19:
fixed by #3043 
I'll open a new issue to track displaying individual sites

closed 2025-02-19 by Williangalvani