bluerobotics/BlueOS #2953: first time running project, repository public.ecr.aws/h4c9p1o0/blueos-cloud-agent-dev not found: name unknown

EstevanPithan, 2024-10-08:
I'm trying to run the project for the first time in Development Environment, but i'm receving this error message, can anyone help me?

~/projetos-blue-os/BlueOS/core/compose$ docker compose pull
WARN[0000] /home/estevan/projetos-blue-os/BlueOS/core/compose/compose.yml: the attribute `version` is obsolete, it will be ignored, please remove it to avoid potential confusion
[+] Pulling 2/2
 ✔ blueos-core Pulled                                                                                                                      1.9s
 ✘ cloud-telemetry Error repository public.ecr.aws/h4c9p1o0/blueos-cloud-agent-dev not found: name unknown: The ...                        2.4s
Error response from daemon: repository public.ecr.aws/h4c9p1o0/blueos-cloud-agent-dev not found: name unknown: The repository with name 'blueos-cloud-agent-dev' does not exist in the registry with id 'h4c9p1o0'

patrickelectric, 2024-10-08:
Hi @EstevanPithan! Do you have a raspberry or any SBC ? This will be the faster way of having it running. The composer is a bit outdated, will try to look into it.

EstevanPithan, 2024-10-08:
Hi Patrick, how are you? I don't have one right now, but a can get one until next week, I think. So the only way right now is with a raspberry or with a drone? I was trying to contribute with Cockpit project