bluerobotics/BlueOS #2962: core: frontend: download extension logs

ES-Alexander, 2024-10-16:
### Current behaviour

Downloading system logs is typically only for BlueOS-core. When troubleshooting it can be relevant to have logs from other software running.

### Expected or desired behaviour

Add an "include extension logs" checkbox to the [System Log Files](https://blueos.cloud/docs/blueos/1.2/advanced-usage/#blueos-settings) section, so the download button includes logs from the `extensions` directory (perhaps specifically any folder that's `extensions/*/logs/`).

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rotu, 2024-10-17:
I don't expect extensions to be able to log to file, especially if the extension fails to start unexpectedly. Maybe we should first add a way to download the existing docker logs with a link on the extensions manager page per #1611.

I had to resort [to telling a user how to manually use the kraken API](https://discuss.bluerobotics.com/t/ping-1d-and-cockpit-standalone-version/17977/17):

> Navigate your web browser to http://blueos.local:9134/v2.0/list_containers and look for something that contains the words “sonarview” and the “name” field. It should be something like (maybe exactly) "/extension-nicknothomsonarview1100".
> 
> If that’s what it is, then go to the following URL (if the name is different, adjust this URL accordingly):
>
> http://blueos-dan.local:9134/v2.0/log?container_name=/extension-nicknothomsonarview1100

JoaoMario109, 2024-10-17:
Reopened it since the PR linked only added download logs for individual extensions in frontend, not closing all requested enhancement

ES-Alexander, 2024-10-17:
> I don't expect extensions to be able to log to file

Per #1640, extensions are expected to have file system access, and can put persistent logs there if they want to.

> Maybe we should first add a way to download the existing docker logs with a link on the extensions manager page

That's what @JoaoMario109's PR just added :-)