bluerobotics/BlueOS #3005: bug: Video stream and Ardupilot

KMonga243, 2024-11-20:
### Bug description

The video streams doesn't work. I'm doing a school project and i want my rov to stream videos but it doesn't recognize my camera i think. And the second one is that i can't see anymore Ardupilote in the autopilot firmware.

### Steps to reproduce

[image]
[image]
[image]

### Primary pain point(s)

To stream a video

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2024-11-20:
Hi @KMonga243, in order to help you, we need the following informations:

1. what camera are you using?
2. are you using a navigator?
3. is this a raspberry pi 3, 4, or 5?

Also, could you **[Download log files from BlueOS services](https://blueos.cloud/docs/1.1/usage/advanced/)** and attach them here so we can check what is happening? 

Thanks

KMonga243, 2024-11-22:
We're using an OV2643 USB Camera, edge navigator and the raspberry 4. Can I have a mail so that i can send the log files because they're too big (33mb) to attach them here.
[Uploading system_logs.zip…]()

joaoantoniocardoso, 2024-11-22:
Right, my address is `joao`, at blue robotics dot com. (Yeah, I'm trying to avoid writing the address it plainly because of spam)

About this specific camera, I couldn't find enough information online, are you sure this camera is capable of providing H264 encoded video stream?

Thanks

KMonga243, 2024-11-25:
system_logs.zip
<https://drive.google.com/file/d/1LTg1_PtiMvpPPP3RqjZd5TvyQCL0cz6e/view?usp=drive_web>
It doesn't provide H264 encoded video but when I ask chatGPT about it, it
says that it provide raw data or compressed like (YUV and JPEG).

KMonga243, 2024-11-25:
So if i understood. Even if i enable the pirate mode i won't see the image in cockpit?

joaoantoniocardoso, 2024-11-26:
> So if i understood. Even if i enable the pirate mode i won't see the image in cockpit?

Correct. Cockpit only works with H264 encoded video streams.

One might say it is possible to encode raw to h264. While it's true,  it's also too resource intensive for a raspberry that also needs to control the vehicle, plus it adds a considerable amount of latency, so it's not really an option.

Maybe you can consider another camera for your project? 

Thanks

patrickelectric, 2025-01-06:
We don't support cameras that does not provide h264 video encode. 
@KMonga243 let us know if you have further questions.

closed 2025-01-06 by patrickelectric