bluerobotics/BlueOS #3029: bug: Can't exit wizard if firmware install fails

vshie, 2024-12-10:
### Bug description

[image]
When setting up a pixhawk based system, the firmware will fail to load. Not a big deal, but when trying to exit the wizard, a page refresh is required

### Steps to reproduce

1.3.1 BlueOS - run wizard with Pixhawk attached to system USB port. 

### Primary pain point(s)

Not intuitive for a new user to deal with!

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2025-01-06:
@vshie did you try more than 2 times ? An abort button should appear.

patrickelectric, 2025-01-08:
Just tested in 1.4 and the abort button shows

patrickelectric, 2025-01-08:
Let us know if the problem persist

closed 2025-01-08 by patrickelectric