bluerobotics/BlueOS #3039: bug: iFrame shows in the other Views on startup

hafizhkarim, 2024-12-24:
### Bug description

Uploading Cockpit 2024-12-24 12-29-00.mp4…

iFrame shows in front of other views

### Steps to reproduce

attatch the iFrame to 1 of the views (not the default views). refresh the page or change to mission view and go back to flight view.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

hafizhkarim, 2024-12-24:
Uploading Cockpit 2024-12-24 12-29-00.mp4…

hafizhkarim, 2024-12-24:
[https://drive.google.com/file/d/18eVlAGHV2GmTm8R-jXDTnvilJvW5_dnt/view?usp=sharing](url)

rafaellehmkuhl, 2024-12-24:
I think this bug is actually on [Cockpit](https://github.com/bluerobotics/cockpit). Could you reopen the issue there?

closed 2024-12-24 by hafizhkarim