bluerobotics/BlueOS #3066: bug: Cannot install OAK-D extension on PC browser terminal

akang-ai, 2025-01-16:
### Bug description

[image]
[image]

### Steps to reproduce

1. Connect to the Internet 2. Open the extension store 3. Install

### Primary pain point(s)

Can't connect to my OAK-D camera, can't get video stream

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

Williangalvani, 2025-02-05:
do you have access to dockerhub?

akang-ai, 2025-02-16:
Yes, I can access dockerhub, so what should I do next, how can I install the OAK-D extension correctly on my Buleos system?

Williangalvani, 2025-02-25:
Hi @akang-ai, were you able to get it working?
I'd suggest trying to update to the latest beta.

qimg412, 2025-03-04:
I have the same problems, but I even could not install any extensions by online store, the network is OK, I can upgrade my Pixhawk, but extension failed.