bluerobotics/BlueOS #3079: bug: IPV4 address lost after a few minutes, restored manually from command line.

rxraine, 2025-01-19:
### Bug description

After a fresh install of 1.4.0-beta.9 on a Raspberry Pi 5 4Gb, everything initially boots up normally. After a couple of minutes, the UI and QGroundControl connections are lost. On the command line, 'ifconfig -a' showed eth0 as having no IPV4 address. Setting this manually with 'sudo ifconfig eth0 192.168.2.2' immediately restores the connections. Rebooting, however, repeats the problem.
Similar issue reported by someone else, only they found that unplugging and replugging the network cable gave them another three minutes before again disconnecting.

### Steps to reproduce

1. Install 1.4.0-beta.9 on a Pi 5 (mine is 4Gb); 8Gb SD Card.
2. Connect with a computer on 192.168.2.1.
3. Open BlueOS and/or QGroundControl.
4. Wait ~three minutes.
Connection to Pi is lost.

### Primary pain point(s)

Need a display and keyboard attached to the Pi to apply the IP address manually.

[system_logs.zip](https://github.com/user-attachments/files/18469157/system_logs.zip)

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2025-01-19:
Hi @rxraine thanks for the report! Can you try beta.10 ? It should fix this problem.

rxraine, 2025-01-19:
Hi @patrickelectric . Thanks for ther quick answer, and on a Sunday too!
Confirmed fixed in beta.10. Wrote image to an SD card, started it up. Left it for over half an hour, no disconnects.

patrickelectric, 2025-01-19:
Great, glad that is working. I'll close for now.

closed 2025-01-19 by patrickelectric