bluerobotics/BlueOS #3161: [Wizard] Duplicated entries for Parameter Sets

joaoantoniocardoso, 2025-02-19:
BlueOS-1.3.1, Raspberry Pi 4

[image]

Autopilot log:
```
2025-02-19 22:25:40.329 | INFO     | logging:callHandlers:1706 - 127.0.0.1:60214 - "GET /v1.0/board HTTP/1.0" 200                                                                                                                             
2025-02-19 22:25:44.953 | DEBUG    | firmware.FirmwareDownload:get_download_url:172 - Got following versions for Vehicle.Sub running Platform.Navigator: ['STABLE-4.5.1', 'STABLE-4.5.0', 'STABLE-4.1.2', 'STABLE-4.1.1', 'STABLE-4.1.0', 'STA
BLE-4.0.3', 'OFFICIAL', 'DEV', 'BETA']                                                                                                                                                                                                        
2025-02-19 22:25:45.079 | DEBUG    | firmware.FirmwareDownload:get_download_url:216 - Downloading following firmware: {'mav-autopilot': 'ARDUPILOTMEGA', 'vehicletype': 'Sub', 'platform': 'navigator', 'git-sha': 'ed138794685eaf43b8fadb1a01
ec429b85d0b398', 'url': 'https://firmware.ardupilot.org/Sub/stable-4.5.1/navigator/ardusub', 'mav-type': 'SUBMARINE', 'mav-firmware-version-type': 'STABLE-4.5.1', 'mav-firmware-version-str': 'V4.5.1', 'latest': 0, 'format': 'ELF', 'mav-fi
rmware-version': '4.5.1', 'mav-firmware-version-major': '4', 'mav-firmware-version-minor': '5', 'mav-firmware-version-patch': '1'}                                                                                                            
2025-02-19 22:25:45.080 | DEBUG    | firmware.FirmwareDownload:get_download_url:172 - Got following versions for Vehicle.Sub running Platform.Navigator: ['STABLE-4.5.1', 'STABLE-4.5.0', 'STABLE-4.1.2', 'STABLE-4.1.1', 'STABLE-4.1.0', 'STA
BLE-4.0.3', 'OFFICIAL', 'DEV', 'BETA']                                                                                                                                                                                                        
2025-02-19 22:25:45.195 | DEBUG    | firmware.FirmwareDownload:get_download_url:216 - Downloading following firmware: {'mav-autopilot': 'ARDUPILOTMEGA', 'vehicletype': 'Sub', 'platform': 'navigator', 'git-sha': '03c12698df56c600f5b3d39f8b
17d414ea6d2a48', 'url': 'https://firmware.ardupilot.org/Sub/stable-4.5.0/navigator/ardusub', 'mav-type': 'SUBMARINE', 'mav-firmware-version-type': 'STABLE-4.5.0', 'mav-firmware-version-str': 'V4.5.0', 'latest': 0, 'format': 'ELF', 'mav-fi
rmware-version': '4.5.0', 'mav-firmware-version-major': '4', 'mav-firmware-version-minor': '5', 'mav-firmware-version-patch': '0'}                                                                                                            
2025-02-19 22:25:45.197 | DEBUG    | firmware.FirmwareDownload:get_download_url:172 - Got following versions for Vehicle.Sub running Platform.Navigator: ['STABLE-4.5.1', 'STABLE-4.5.0', 'STABLE-4.1.2', 'STABLE-4.1.1', 'STABLE-4.1.0', 'STA
BLE-4.0.3', 'OFFICIAL', 'DEV', 'BETA']                                                                                                                                                                                                        
2025-02-19 22:25:45.319 | DEBUG    | firmware.FirmwareDownload:get_download_url:216 - Downloading following firmware: {'mav-autopilot': 'ARDUPILOTMEGA', 'vehicletype': 'Sub', 'platform': 'navigator', 'git-sha': '2dd0bb7d4c85ac48437f139d66
df648fc0e1d4ae', 'url': 'https://firmware.ardupilot.org/Sub/stable-4.1.2/navigator/ardusub', 'mav-type': 'SUBMARINE', 'mav-firmware-version-type': 'STABLE-4.1.2', 'mav-firmware-version-str': 'V4.1.2', 'latest': 0, 'format': 'ELF', 'mav-fi
rmware-version': '4.1.2', 'mav-firmware-version-major': '4', 'mav-firmware-version-minor': '1', 'mav-firmware-version-patch': '2'}                                                                                                            
2025-02-19 22:25:45.320 | DEBUG    | firmware.FirmwareDownload:get_download_url:172 - Got following versions for Vehicle.Sub running Platform.Navigator: ['STABLE-4.5.1', 'STABLE-4.5.0', 'STABLE-4.1.2', 'STABLE-4.1.1', 'STABLE-4.1.0', 'STA
BLE-4.0.3', 'OFFICIAL', 'DEV', 'BETA']                                                                                                                                                                                                        
2025-02-19 22:25:45.443 | DEBUG    | firmware.FirmwareDownload:get_download_url:216 - Downloading following firmware: {'mav-autopilot': 'ARDUPILOTMEGA', 'vehicletype': 'Sub', 'platform': 'navigator', 'git-sha': '8253c7e50630708c601ebe11b6
45b8a06e2183ce', 'url': 'https://firmware.ardupilot.org/Sub/stable-4.1.1/navigator/ardusub', 'mav-type': 'SUBMARINE', 'mav-firmware-version-type': 'STABLE-4.1.1', 'mav-firmware-version-str': 'V4.1.1', 'latest': 0, 'format': 'ELF', 'mav-fi
rmware-version': '4.1.1', 'mav-firmware-version-major': '4', 'mav-firmware-version-minor': '1', 'mav-firmware-version-patch': '1'}                                                                                                            
2025-02-19 22:25:45.444 | DEBUG    | firmware.FirmwareDownload:get_download_url:172 - Got following versions for Vehicle.Sub running Platform.Navigator: ['STABLE-4.5.1', 'STABLE-4.5.0', 'STABLE-4.1.2', 'STABLE-4.1.1', 'STABLE-4.1.0', 'STA
BLE-4.0.3', 'OFFICIAL', 'DEV', 'BETA']                                                                                                                                                                                                        
2025-02-19 22:25:45.542 | DEBUG    | firmware.FirmwareDownload:get_download_url:216 - Downloading following firmware: {'mav-autopilot': 'ARDUPILOTMEGA', 'vehicletype': 'Sub', 'platform': 'navigator', 'git-sha': 'f2af3c7ed2907be914c41d8512
654a77498d3870', 'url': 'https://firmware.ardupilot.org/Sub/stable-4.1.0/navigator/ardusub', 'mav-type': 'SUBMARINE', 'mav-firmware-version-type': 'STABLE-4.1.0', 'mav-firmware-version-str': 'V4.1.0', 'latest': 0, 'format': 'ELF', 'mav-fi
rmware-version': '4.1.0', 'mav-firmware-version-major': '4', 'mav-firmware-version-minor': '1', 'mav-firmware-version-patch': '0'}                                                                                                            
2025-02-19 22:25:45.544 | DEBUG    | firmware.FirmwareDownload:get_download_url:172 - Got following versions for Vehicle.Sub running Platform.Navigator: ['STABLE-4.5.1', 'STABLE-4.5.0', 'STABLE-4.1.2', 'STABLE-4.1.1', 'STABLE-4.1.0', 'STA
BLE-4.0.3', 'OFFICIAL', 'DEV', 'BETA']                                                                                                                                                                                                        
2025-02-19 22:25:45.636 | DEBUG    | firmware.FirmwareDownload:get_download_url:216 - Downloading following firmware: {'mav-autopilot': 'ARDUPILOTMEGA', 'vehicletype': 'Sub', 'platform': 'navigator', 'git-sha': '96882ed0aeeb822dd80575312b
782fe47c352b25', 'url': 'https://firmware.ardupilot.org/Sub/stable-4.0.3/navigator/ardusub', 'mav-type': 'SUBMARINE', 'mav-firmware-version-type': 'STABLE-4.0.3', 'mav-firmware-version-str': 'V4.0.3', 'latest': 0, 'format': 'ELF', 'mav-fi
rmware-version': '4.0.3', 'mav-firmware-version-major': '4', 'mav-firmware-version-minor': '0', 'mav-firmware-version-patch': '3'}                                                                                                            
2025-02-19 22:25:45.637 | DEBUG    | firmware.FirmwareDownload:get_download_url:172 - Got following versions for Vehicle.Sub running Platform.Navigator: ['STABLE-4.5.1', 'STABLE-4.5.0', 'STABLE-4.1.2', 'STABLE-4.1.1', 'STABLE-4.1.0', 'STA
BLE-4.0.3', 'OFFICIAL', 'DEV', 'BETA']                                                                                                                                                                                                        
2025-02-19 22:25:45.729 | DEBUG    | firmware.FirmwareDownload:get_download_url:216 - Downloading following firmware: {'mav-autopilot': 'ARDUPILOTMEGA', 'vehicletype': 'Sub', 'platform': 'navigator', 'git-sha': 'ed138794685eaf43b8fadb1a01
ec429b85d0b398', 'url': 'https://firmware.ardupilot.org/Sub/stable/navigator/ardusub', 'mav-type': 'SUBMARINE', 'mav-firmware-version-type': 'OFFICIAL', 'mav-firmware-version-str': 'V4.5.1', 'latest': 0, 'format': 'ELF', 'mav-firmware-ver
[... cut ...]
2025-02-19 22:25:45.942 | INFO     | logging:callHandlers:1706 - 127.0.0.1:60216 - "GET /v1.0/available_firmwares?vehicle=Sub HTTP/1.0" 200
2025-02-19 22:25:45.948 | INFO     | logging:callHandlers:1706 - 127.0.0.1:60218 - "GET /v1.0/available_firmwares?vehicle=Sub HTTP/1.0" 200
2025-02-19 22:25:46.665 | INFO     | logging:callHandlers:1706 - 127.0.0.1:60250 - "GET /v1.0/board HTTP/1.0" 200
2025-02-19 22:25:50.366 | INFO     | logging:callHandlers:1706 - 127.0.0.1:60270 - "GET /v1.0/board HTTP/1.0" 200
```

Williangalvani, 2025-02-20:
this should have been fixed on master by https://github.com/bluerobotics/BlueOS/pull/3069

joaoantoniocardoso, 2025-02-20:
> this should have been fixed on master by #3069

makes sense to fix that, I'll close it for now, then

closed 2025-02-20 by joaoantoniocardoso