bluerobotics/BlueOS #3169: BlueOS cannot be installed from .img file on RPI4 CM

rmackay9, 2025-02-21:
### Current behaviour

I've recently been working with @Williangalvani to install BlueOS on an aerial photography quadcopter which has an RPI4 CM as it's companion computer.  The BlueOS image files ([found here](https://github.com/bluerobotics/BlueOS/releases)) cannot be used to install BlueOS.  If they are used the RPI refuses to boot and displays, "This board requires newer software" on the display.  Instead the [manual install instructions](https://github.com/bluerobotics/BlueOS/tree/master/install) must be used.

The fail to boot error message shown is below
[image]

It would be much faster and easiert to setup BlueOS on an RPI4 CM if the regular images worked

### Expected or desired behaviour

I'm using [this RPI4 compute module](https://www.digikey.jp/en/products/detail/raspberry-pi/SC0678/13530917) mounted on the [standard RPI4 CM I/O board](https://www.raspberrypi.com/products/compute-module-4-io-board/).
[RPI4 install instructions are her](https://www.raspberrypi.com/documentation/computers/compute-module.html#flash-compute-module-emmc)e

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

rmackay9, 2025-04-21:
By the way, my RPI4 CM uses the eMMC

Williangalvani, 2025-04-21:
@randy I managed to install it on a cm4's emmc by using the latest 1.4.0 release.
I used the [BlueOS-raspberry-linux-arm64-v8-bookworm-pi5.zip](https://github.com/bluerobotics/BlueOS/releases/download/1.4.0/BlueOS-raspberry-linux-arm64-v8-bookworm-pi5.zip) image as I wanted a 64bits os. but I expect the armv7 bookworm image would also work.

rmackay9, 2025-04-22:
Great, I was aiming to test today but didn't quite get to it.  Surely tomorrow.

closed 2025-04-23 by rmackay9