bluerobotics/BlueOS #3170: Wifi networks cannot be seen on RPI4 CM

rmackay9, 2025-02-21:
### Bug description

It's slightly hard to believe and perhaps I'm doing something wrong but it seems like on an RPI4 CM wifi networks cannot be seen after clicking the Wifi icon on the top right

[image]

Wifi networks certainly are available though and I can see them with my regular PC

[image]

The RPI4 CM also does have an antenna attached although I suppose I should admit that I borrowed it from an [ESP32 C3](https://www.seeedstudio.com/Seeed-Studio-XIAO-ESP32C3-Pre-Soldered-p-6331.html) telemetry radio.

[image]

### Steps to reproduce

1. power on RPI4 CM running BlueOS and ensure connected via LAN cable to local ethernet
2. on PC, open a chrome browser and enter URL http://blueos-avahi.local
3. from top-right menu bar select the wifi icon
4. note that "no wifi networks available"

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

Williangalvani, 2025-03-06:
that seems to be caused by wpa_supplicant crashing when we update it's wpa_supplicant.service file.
The new file points to /et/wpa_supplicant/wpa_supplicant.conf. which is not being populated outside of our deployment script. We should validate that the file exists on blueos_startup_update.py

rmackay9, 2025-03-06:
Thanks so much for looking into this!

rmackay9, 2025-04-23:
After testing the latest 1.4 version I'm not seeing this problem any more, thanks!

closed 2025-04-23 by rmackay9