bluerobotics/BlueOS #3171: bug: RTSP Video Play in cockpit GCS

zshun-robot, 2025-02-21:
### Bug description

The SIYI FPV RTSP video stream cannot be played, I can use VLC to view the video stream, but it cannot be played in BlueOS. The fake source ball work fine, but my rtsp IPC camera can not work, I want play video in cockpit GCS.

### Steps to reproduce

In video manager page, Redirect source add RTSP: rtsp://192.168.144.26:8554/main.264, it can't set Encoding, available sizes or available framerates. Could it be that this is the reason for the failure to get the video stream?

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2025-02-21:
Hi, 

This looks like a networking configuration issue: for an easy setup, the cameras and the vehicle should always live in the same subnet. This means that if your camera is at 192.168.144.26, your vehicle IP must be between 192.168.144.1 and 192.168.144.255.

The idea behind this is that the RTSP packets must go to BlueOS, and then BlueOS will send it to your computer via Cockpit.

zshun-robot, 2025-03-02:
I have resolved this issue. The reason was that my camera's default encoding is H.265, while Cockpit only supports H.264 encoding.

joaoantoniocardoso, 2025-03-02:
> I have resolved this issue. The reason was that my camera's default encoding is H.265, while Cockpit only supports H.264 encoding.

Thanks for letting us know!

Cockpit actually supports H265 if the browser + operating system supports it (the latest Safari, for example)

We were talking last Cockpit meeting (which is public btw) about giving some feedback for the user when the stream is h265 and the browser doesn't support it.