bluerobotics/BlueOS #3191: Issues from R&D meeting 06/03/25

patrickelectric, 2025-03-06:
- RTSP streams being forgotten #3187 
- When uploading fw into firmware page:
    - Select firmware, click again in fw selection box, choose nothing, do upload
- Camera gimbal configuration should restart vehicle
- Parameters smallers than 0.00001 shows as zero
- Camera streams appears to be lost with hot-plugging camera
- Rename SERVO1_FUNCTION into setup page to output1 and etc
    - Highlight the pwm pins of navigator or pixhawk as well

patrickelectric, 2025-03-06:
@joaoantoniocardoso can you double check if hot-plugging is a problem with latest beta ?

joaoantoniocardoso, 2025-03-07:
> @joaoantoniocardoso can you double check if hot-plugging is a problem with latest beta ?

Which camera?

patrickelectric, 2025-03-07:
> > [@joaoantoniocardoso](https://github.com/joaoantoniocardoso?rgh-link-date=2025-03-07T01%3A02%3A08.000Z) can you double check if hot-plugging is a problem with latest beta ?
> 
> Which camera?

usb

patrickelectric, 2025-03-18:
I did some tests and it appears that the usb is working fine

joaoantoniocardoso, 2025-03-18:
> I did some tests and it appears that the usb is working fine

Yep, I failed to find any problems here. I even managed to plug the USB connector in a way that it was automatically constantly disconnecting hahahah