bluerobotics/BlueOS #3195: bug: Internal Server Error while uploading custom ArduPilot Firmware

Michael-F-Bryan, 2025-03-11:
### Bug description

We are trying to upload some custom ArduPilot firmware to a CUAV-7-Nano autopilot using the firmware upload page built into BlueOS, however every time we try to do the firmware upload it fails with an *Internal Server Error*.

### Steps to reproduce

To reproduce the issue, go through the normal firmware upload workflow.

1. Go to the *Autopilot Firmware* tab
2. Expand the *Firmware Update* widget
3. Click *Upload firmware from file*
4. Select your firmware file (see attached)
5. Click the *Install Firmware* button
6. Wait a couple of seconds and you'll get an *Internal Server Error* popup

https://github.com/user-attachments/assets/c9a5e66e-8d75-4901-9c44-6e26c00e0e1d

To help with reproducing, here is my exact firmware binary: 

[ardusub-firmware.zip](https://github.com/user-attachments/files/19185078/ardusub-firmware.zip)

The only difference between that firmware and the main ardupilot repo is that we've added a custom frame file. The exact commit is https://github.com/sunfish-robotics/ardupilot/commit/19ae91d244951bc5bd83e067bb4170161c87ab18, and once you've got the dev environment set up, you can build it using `make sunfish`.

### Primary pain point(s)

We are encountering this on an autonomous underwater vehicle where the only way to upload new ardupilot firmware is via BlueOS. If we can't use the autopilot firmware manager for updating firmware, it means we'd need physical access to the device... which isn't possible when it's in a sealed container at the bottom of the ocean. 

### Additional context

I managed to look through the logs, and it seems like the exception is triggered from here:

https://github.com/bluerobotics/BlueOS/blob/028ce715e7552a1cd22664b6f5863dbd5fe7a067/core/services/ardupilot_manager/firmware/FirmwareManagement.py#L107

<details>
<summary>(click to see the full stack trace)</summary>

```
2025-03-11 13:31:26.164 | INFO     | logging:callHandlers:1706 - 127.0.0.1:46056 - "POST /v1.0/install_firmware_from_file?board_name=CUAV-7-Nano HTTP/1.0" 500
2025-03-11 13:31:26.165 | ERROR    | logging:callHandlers:1706 - Exception in ASGI application

Traceback (most recent call last):

  File "/home/pi/services/ardupilot_manager/main.py", line 46, in <module>
    loop.run_until_complete(server.serve())
    │    │                  │      └ <function Server.serve at 0xf6ddb708>
    │    │                  └ <uvicorn.server.Server object at 0xf624c470>
    │    └ <function BaseEventLoop.run_until_complete at 0xf71c5078>
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>

  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 640, in run_until_complete
    self.run_forever()
    │    └ <function BaseEventLoop.run_forever at 0xf71c5028>
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 607, in run_forever
    self._run_once()
    │    └ <function BaseEventLoop._run_once at 0xf71c5f28>
    └ <_UnixSelectorEventLoop running=True closed=False debug=False>
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1922, in _run_once
    handle._run()
    │      └ <function Handle._run at 0xf7264848>
    └ <Handle Task.task_wakeup(<Future finished result=None>)>
  File "/usr/local/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
    │    │            │    │           │    └ <member '_args' of 'Handle' objects>
    │    │            │    │           └ <Handle Task.task_wakeup(<Future finished result=None>)>
    │    │            │    └ <member '_callback' of 'Handle' objects>
    │    │            └ <Handle Task.task_wakeup(<Future finished result=None>)>
    │    └ <member '_context' of 'Handle' objects>
    └ <Handle Task.task_wakeup(<Future finished result=None>)>
> File "/usr/local/lib/python3.11/site-packages/uvicorn/protocols/http/h11_impl.py", line 396, in run_asgi
[... 156 lines cut ...]
    └ <autopilot_manager.AutoPilotManager object at 0xf652daf0>

  File "/home/pi/services/ardupilot_manager/autopilot_manager.py", line 652, in install_firmware_from_file
    self.firmware_manager.install_firmware_from_file(firmware_path, board, default_parameters)
    │    │                │                          │              │      └ None
    │    │                │                          │              └ FlightController(name='CUAV-7-Nano', manufacturer='ArduPilot', platform=<Platform.GenericSerial: 'GenericSerial'>, path='/dev...
    │    │                │                          └ PosixPath('/root/.config/ardupilot-manager/firmware/custom_firmware')
    │    │                └ <function FirmwareManager.install_firmware_from_file at 0xf660b438>
    │    └ <firmware.FirmwareManagement.FirmwareManager object at 0xf36fa610>
    └ <autopilot_manager.AutoPilotManager object at 0xf652daf0>

  File "/home/pi/services/ardupilot_manager/firmware/FirmwareManagement.py", line 112, in install_firmware_from_file
    raise FirmwareInstallFail("Could not install firmware.") from error
          └ <class 'exceptions.FirmwareInstallFail'>

exceptions.FirmwareInstallFail: Could not install firmware.
```

</details>

Unfortunately, the code does `raise FirmwareInstallFail("Could not install firmware.") from error` and the logs aren't printing the original exception, so I wasn't able to troubleshoot any further.

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

mc-nugget, 2025-04-01:
As of 1.4.0-beta.18 the error is now slightly different but the pain point is the same.

[image]

ilia-enginer, 2025-10-03:
Bug description
We are trying to upload some custom ArduPilot firmware to a speedybeef4v3 autopilot using the firmware upload page built into BlueOS, however every time we try to do the firmware upload it fails. error: file_firmware_install_fail. I am using the latest stable version of blueos.

Steps to reproduce
To reproduce the issue, go through the normal firmware upload workflow.

Go to the Autopilot Firmware
Expand the Firmware Update
Click Upload firmware from file
Select your firmware file (see attached)
Click the Install Firmware
Wait a couple of seconds and you'll get an Error. error: file_firmware_install_fail.

[image]

closed 2026-10-14 by patrickelectric