bluerobotics/BlueOS #3242: [Wifi] Error creating hotspot

joaoantoniocardoso, 2025-03-31:
Running 1.4.0-beta.19

[image]

Relevant log section:

```python
2025-03-31 21:28:41.540 | INFO     | wifi_handlers.wpa_supplicant.Hotspot:stop:188 - Stopping hotspot.
2025-03-31 21:28:41.542 | INFO     | commonwealth.utils.DHCPServerManager:stop:133 - Tried to stop DHCP Server, but it was already not running.
2025-03-31 21:28:43.176 | INFO     | logging:callHandlers:1706 - 127.0.0.1:44740 - "GET /v1.0/saved HTTP/1.0" 200
2025-03-31 21:28:43.242 | INFO     | logging:callHandlers:1706 - 127.0.0.1:44742 - "GET /v1.0/hotspot_credentials HTTP/1.0" 200
2025-03-31 21:28:43.569 | INFO     | __main__:network_status:59 - ------------------  ------------------------------------
2025-03-31 21:28:43.572 | INFO     | __main__:network_status:59 - bssid               4e:6c:94:04:46:20
2025-03-31 21:28:43.574 | INFO     | __main__:network_status:59 - freq                2462
2025-03-31 21:28:43.575 | INFO     | __main__:network_status:59 - ssid                Zenite
2025-03-31 21:28:43.576 | INFO     | __main__:network_status:59 - id                  0
2025-03-31 21:28:43.577 | INFO     | __main__:network_status:59 - mode                station
2025-03-31 21:28:43.584 | INFO     | __main__:network_status:59 - wifi_generation     4
2025-03-31 21:28:43.586 | INFO     | __main__:network_status:59 - pairwise_cipher     CCMP
2025-03-31 21:28:43.588 | INFO     | __main__:network_status:59 - group_cipher        CCMP
2025-03-31 21:28:43.589 | INFO     | __main__:network_status:59 - key_mgmt            WPA2-PSK
2025-03-31 21:28:43.590 | INFO     | __main__:network_status:59 - wpa_state           COMPLETED
2025-03-31 21:28:43.591 | INFO     | __main__:network_status:59 - ip_address
2025-03-31 21:28:43.593 | INFO     | __main__:network_status:59 - p2p_device_address  ba:27:eb:0f:d6:15
2025-03-31 21:28:43.594 | INFO     | __main__:network_status:59 - address             b8:27:eb:0f:d6:15
2025-03-31 21:28:43.595 | INFO     | __main__:network_status:59 - uuid                adc05bf8-0554-5bdb-b209-4c806336af26
2025-03-31 21:28:43.596 | INFO     | __main__:network_status:59 - ieee80211ac
2025-03-31 21:28:43.597 | INFO     | __main__:network_status:59 - state
2025-03-31 21:28:43.600 | INFO     | __main__:network_status:59 - disabled
2025-03-31 21:28:43.602 | INFO     | __main__:network_status:59 - ------------------  ------------------------------------
2025-03-31 21:28:43.608 | INFO     | logging:callHandlers:1706 - 127.0.0.1:44746 - "GET /v1.0/status HTTP/1.0" 200
2025-03-31 21:28:43.640 | INFO     | wifi_handlers.wpa_supplicant.Hotspot:start:167 - Starting hotspot.
2025-03-31 21:28:43.642 | INFO     | wifi_handlers.wpa_supplicant.Hotspot:_create_temp_config_file:243 - Saving temporary hostapd config file on /hostapd.conf
2025-03-31 21:28:45.653 | DEBUG    | wifi_handlers.wpa_supplicant.Hotspot:_create_virtual_interface:129 - Deleting virtual access point interface (if exists).
2025-03-31 21:28:46.647 | DEBUG    | wifi_handlers.wpa_supplicant.Hotspot:_create_virtual_interface:139 - Creating virtual access point interface.
2025-03-31 21:28:47.056 | DEBUG    | wifi_handlers.wpa_supplicant.Hotspot:_create_virtual_interface:152 - Starting virtual access point interface.
uap0: interface state UNINITIALIZED->ENABLED
uap0: AP-ENABLED
2025-03-31 21:28:50.370 | INFO     | commonwealth.utils.DHCPServerManager:stop:133 - Tried to stop DHCP Server, but it was already not running.

dnsmasq: unknown interface uap0
2025-03-31 21:28:53.374 | ERROR    | commonwealth.utils.apis:custom_route_handler:36 - Unhandled service exception.
2025-03-31 21:28:53.375 | ERROR    | commonwealth.utils.apis:custom_route_handler:37 - Unable to start hotspot. Unable to start DHCP Server.
Traceback (most recent call last):

  File "/home/pi/libs/commonwealth/commonwealth/utils/DHCPServerManager.py", line 122, in start
    raise RuntimeError(f"Failed to initialize Dnsmasq ({exit_code}).")

RuntimeError: Failed to initialize Dnsmasq (2).

The above exception was the direct cause of the following exception:

Traceback (most recent call last):

  File "/home/pi/services/wifi/wifi_handlers/wpa_supplicant/Hotspot.py", line 183, in start
    await self._dhcp_server.restart()
          │    │            └ <function Dnsmasq.restart at 0x75570938>
          │    └ <commonwealth.utils.DHCPServerManager.Dnsmasq object at 0x74fcff30>
          └ <wifi_handlers.wpa_supplicant.Hotspot.HotspotManager object at 0x74fc4c50>

  File "/home/pi/libs/commonwealth/commonwealth/utils/DHCPServerManager.py", line 137, in restart
    await self.start()
          │    └ <function Dnsmasq.start at 0x75570898>
          └ <commonwealth.utils.DHCPServerManager.Dnsmasq object at 0x74fcff30>

  File "/home/pi/libs/commonwealth/commonwealth/utils/DHCPServerManager.py", line 125, in start
    raise RuntimeError("Unable to start DHCP Server.") from error

RuntimeError: Unable to start DHCP Server.

[... 171 lines cut ...]
          └ <wifi_handlers.wpa_supplicant.WifiManager.WifiManager object at 0x755588d0>

  File "/home/pi/services/wifi/wifi_handlers/wpa_supplicant/WifiManager.py", line 83, in try_connect_to_network
    await self.connect_to_network(network_id, timeout=40)
          │    │                  └ 0
          │    └ <function WifiManager.connect_to_network at 0x7501f758>
          └ <wifi_handlers.wpa_supplicant.WifiManager.WifiManager object at 0x755588d0>

  File "/home/pi/services/wifi/wifi_handlers/wpa_supplicant/WifiManager.py", line 331, in connect_to_network
    await self.enable_hotspot(save_settings=False)
          │    └ <function WifiManager.enable_hotspot at 0x7501fac8>
          └ <wifi_handlers.wpa_supplicant.WifiManager.WifiManager object at 0x755588d0>

  File "/home/pi/services/wifi/wifi_handlers/wpa_supplicant/WifiManager.py", line 496, in enable_hotspot
    await self.hotspot.start()
          │    └ <property object at 0x7554e910>
          └ <wifi_handlers.wpa_supplicant.WifiManager.WifiManager object at 0x755588d0>

  File "/home/pi/services/wifi/wifi_handlers/wpa_supplicant/Hotspot.py", line 185, in start
    raise RuntimeError(f"Unable to start hotspot. {error}") from error

RuntimeError: Unable to start hotspot. Unable to start DHCP Server.
2025-03-31 21:28:53.570 | INFO     | logging:callHandlers:1706 - 127.0.0.1:44726 - "POST /v1.0/connect?hidden=false HTTP/1.0" 500
2025-03-31 21:28:53.582 | DEBUG    | wifi_handlers.wpa_supplicant.WifiManager:auto_reconnect:442 - Regained connection.
```

patrickelectric, 2025-03-31:
Can you provide the steps to replicate the error ?

joaoantoniocardoso, 2025-03-31:
I just started it and connected to my wifi network, which might not be exactly the necessary steps.

joaoantoniocardoso, 2025-03-31:
since it is related to dnsmasq, it might have relationship to the cable-guy, which also uses it. Here it's using a backup server to the static IP 192.168.2.2 (which is the default configuration for 1.4.0)

joaoantoniocardoso, 2025-03-31:
[image]

PR #3244 (2025-04-01): core:libs:commonwealth: Change to bind-dynamic
Fix #3242

This allows `dnsmasq` to run even though the interface is not totally configured or up.

## Summary by Sourcery

Enhancements:
- Modify DHCP server binding strategy to use dynamic interface binding instead of strict interface binding

## Summary by Sourcery

Modify dnsmasq DHCP server binding strategy to use dynamic interface binding

Bug Fixes:
- Resolve issue #3242 by allowing dnsmasq to run even when the network interface is not fully configured

Enhancements:
- Change dnsmasq configuration from strict interface binding to dynamic in [... cut]
closed 2025-04-01 by patrickelectric