bluerobotics/BlueOS #3253: [Kraken] Slowly growing RAM usage

joaoantoniocardoso, 2025-04-04:
I have a Pi3 (BlueOS 1.4.0-beta.20) with an `htop` session running on an attached 7" HDMI monitor. I'm not interacting with this Pi3, it's sitting idle with a Pixhawk and a BR camera connected.

Yesterday, I noticed that kraken was using ~9% of RAM, which caught my attention because it was the highest among the processes, and I started to check on it from time to time. Today morning the RAM was at ~12%, and now it's ~15%, so I started tracking it, below data are 8h of observation after 48h of uptime:

[image]

joaoantoniocardoso, 2025-04-04:
Update from @joaomariolago's investigation:

He worked on an accelerated kraken (by simply changing the frequency of the loops), and by isolating parts of the code, he narrowed down the leak to the following section of the code:

[image]

Which means it's coming from the `harbor` library.

Williangalvani, 2025-04-04:
we don't actually need to do this cleanup every 60s, right? 
we could do it only once per run and that would be fine...

joaomariolago, 2025-04-04:
> we don't actually need to do this cleanup every 60s, right? we could do it only once per run and that would be fine...

The cleanup loop can run at lower speeds but it's good for some cases where containers can be keept alive. But for better clarification the code that is probably generating the leak is the following:

```python
    @staticmethod
    async def get_running_containers() -> List[ContainerModel]:
        async with DockerCtx() as client:
            containers = await client.containers.list(filters={"status": ["running"]})  # type: ignore

            return [
                ContainerModel(
                    name=container["Names"][0],
                    image=container["Image"],
                    image_id=container["ImageID"],
                    status=container["Status"],
                )
                for container in containers
            ]
```

This code is used in other parts but as more notable for the leak on the `init_dead_extensions` task that is important for **kraken** and runs with a period of `5s`

joaoantoniocardoso, 2025-04-11:
The same kind of leak happens with version-chooser, but it's even slower there.

joaoantoniocardoso, 2025-04-14:
- After approximately 9 days, the swap (1024Mb) was full.
- After 10 days, 7 hours, and 19 minutes, the system automatically restarted. Below are the recorded data and Kraken logs:

[image]

[image]

[kraken-20250405-015908.gz](https://github.com/user-attachments/files/19735577/kraken-20250405-015908.gz)