bluerobotics/BlueOS #3262: bug: MAVLink Enpoints disabled/enabled slider is misleading

rmackay9, 2025-04-09:
### Bug description

The MAVLink Endpoints screen has a disabled/enabled slider and when the user changes its position from enabled to disabled, the screen updates but it then misleadingly appears that the mavlink connection is disabled when it is not.  It is actually still enabled and active.  I know its still active because the BlueOS heartbeat icon shows the connection (to the autopilot in my case) is still active, cockpit still works as well

Other fields do seem to take effect immediately including the port number

### Steps to reproduce

In my setup I found this issue by connecting a CubeOrangePlus to the RPI4 CM running BlueOS.  I then created a new MAVLink endpoint like this:

AutoPilot Client Link, UDP Client, 192.168.144.15:14560

Whether the slider was enabled or disabled, the connection to the autopilot remained.  If I changed the port to 14561 then BlueOS/Cockpit could no longer connect

### Primary pain point(s)

It's misleading which meant that while investigating a connection issue, I was misled until eventually finding the issue was in the display itself

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2025-04-09:
Hi, I believe this is was solved in 1.4.0-beta.21, the enable/disable switch was only taking effect on mavrouter, not the others.

rmackay9, 2025-05-02:
I think this is fixed in the 1.4 release which is now out so closing, thanks!

closed 2025-05-02 by rmackay9