bluerobotics/BlueOS #3289: bug: MAVLink Endpoint screen empty on new install of 1.4

rmackay9, 2025-04-30:
### Bug description

I've recently installed the latest release, 1.4 on an RPI4 CM and found that unlike in earlier versions, the MAVLink Endpoints screen is completely blank

### Steps to reproduce

The method I used to install BlueOS 1.4 is below

1. on my Ubuntu 24 machine:
    - sudo apt install rpiboot
    - sudo apt install rpi-imager
2. connect RPI4 CM to RPI I/O board
   - connected USB to PC
   - added jumper so it doesn't boot, "Fit jumper to disable eMMC Boot
   - powered up RPI
3. On my Ubuntu PC
    - run rpiboot
    - run rpi-imager
        - Choose Device: Raspberry Pi 4
        - Operating System: Use custom, select downloaded BlueOS .img file
        - Storage: select RPI drive (should have appeared after rpiboot was run)
        - Select "No" when asked to apply special settings

4. Reinstall RPI4 CM on Ochin carrier board

5. Connect PC and RPI4 CM to the same ethernet subnet
    - Open web browser with URL, http://192.168.2.2

6. Use wizard to connect to:
    a. local Wifi network
    b. select "Other vehicle setup"
        - Vehicle Name: RandysCopter
        - MDNS Name: "blueos" (default)

7. RPI4 / BlueOS Settings:
    - pirate mode from top right
    - select networking icon,
        a. select "eth0"
        b. ADD STATIC IP 192.168.144.20
        c. "DISABLE DHCP SERVER"
        d. beside "192.168.2.2 Static IP" push garbage can icon
    - reconnect with web browser using URL, http://192.168.144.20
    - enable pirate mode again by pushing on buddha on top right (?)

8. Check MAVLink Endpoints screen -- it's blank!
9. 

### Primary pain point(s)

This means that I need to setup all mavlink endpoints manually which is time consuming

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

Williangalvani, 2025-04-30:
hmmm it seems we don't handle the case of "No flight controller detected" correctly on the mavlink endpoints page.
they should show up if one is detect or you switch to SITL

[image]

rmackay9, 2025-06-03:
This is fixed now that we have the manual flight controller option in 1.5.0, thanks!

closed 2025-06-03 by rmackay9