bluerobotics/BlueOS #3306: bug: CPU 99% until MAVLink-Server restarted

rmackay9, 2025-05-14:
### Bug description

It seems that at least on my setup, the RPI4 CM's CPU is running at nearly full CPU until I restart MAVLink-Server from the MAVLink Endpoints screen (I do this by temporarily selecting MAVLinkRouter and then MAVLink-Server again)

[image]

FYI @Williangalvani

### Steps to reproduce

Install stable BlueOS (1.4?), connect to an autopilot over Ethernet.  Ensure the RPI/BlueOS can connect to the local Wifi.

From a PC:

- open a chrome browser and connect to BlueOS using http://blueos-avahi.local/
- enter Pirate mode
- open the System Information screen and check the CPU usage of "mavlink-server"

To resolve the high CPU:

- open the MAVLink Endpoint screen
- push the "MAVLinkRouter" option
- push the "MAVLink-Server" option
- return to the System Information screen and confirm the CPU usage of "mavlink-server" has dropped

### Primary pain point(s)

With the RPI using 100% CPU it becomes sluggish and hot

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

joaoantoniocardoso, 2025-05-14:
Hi @rmackay9, thanks for reporting it! Could you send us the BlueOS logs?

rmackay9, 2025-05-16:
Hi @joaoantoniocardoso,

Thanks for having a look at this.  [The logs are here](https://www.dropbox.com/scl/fi/vd917r8c42oqj8z2uddo6/randy-system_logs-16May2025.zip?rlkey=bewpsujtu8qctkek6m183g4g1&dl=0).

By the way, while switching to MAVLinkRouter resolves the CPU issue, it causes BlueOS to be unable to connect to the autopilot so this is a bit of a show stopper for me.  I suspect that I may have a mistake in my mavlink endpoints but I'm not sure.

Thanks again!

joaoantoniocardoso, 2025-05-16:
It looks like the MS is throttling on the default Zenoh endpoint.

Would you mind telling us what is connected to the added endpoints?

- UdpServer "0.0.0.0:14560"
- UdpClient "192.168.144.15:14560"
- UdpClient "192.168.144.20:14550"

joaoantoniocardoso, 2025-05-16:
Hi @rmackay9,

I had a meeting with @Williangalvani, and we detected a loop in your settings:

BlueOS, by default, has the endpoint `GCS Server Link` as a **UDP Server** at port `15500`, and you've configured an additional endpoint as a **UDP Client** to its own IP address (`192.168.144.20`, from your `eth0` interface), to the same port `15550`.

This connection creates a loopback that fries the whole Mavlink Server.

Let us know whether the CPU usage goes back down when you disable that UDP Client?

Thanks!

rmackay9, 2025-05-19:
Hi @joaoantoniocardoso thanks very much for looking into this.  Here's what I see on the MAVLink Endpoints screen and Autopilot Firmware screen (which is more of an Autopilot MAVLink endpoint I guess) 

[image]
[image]

So I think you're saying that the "GCS Client Link" is the problem but it's disabled.  If I completely remove the item and reboot the RPI then the CPU drops to 7%, thanks!

I guess maybe this is just a re-occurance of https://github.com/bluerobotics/BlueOS/issues/3262 but it's not included in the BlueOS version that I'm running.

[image]

Sorry about that but thanks again for your help!

closed 2025-05-19 by rmackay9