bluerobotics/BlueOS #3415: bug: Video Streams's Redirect source not working in 1.5.0-beta8?

rmackay9, 2025-07-08:
### Bug description

I can't seem to get the Video Manager in 1.5.0-beta8 to work.  I'm unsure if I've messed up some configuration or if the latest beta has broken it but I'm sure I had it working about a month ago when I wrote [this AP blog post](https://discuss.ardupilot.org/t/ardupilot-and-blueos-for-companion-computers/134879) (search for "Setup Cockpit Video Streaming")

Now the Video Manager always says "Not running" and the little thumbnail is not appearing
[image]

### Steps to reproduce

Install BlueOS 1.5.0-beta8 on an RPI5 as described in [this AP blog post](https://discuss.ardupilot.org/t/ardupilot-and-blueos-for-companion-computers/134879)
Connect a Siyi A8 to the RPI5 via Ethernet
Set the RPI5's IP address to 192.168.144.20
Open BlueOS's Video Streams:
    - select, “ADD STREAM”
    - Stream nickname: “Siyi A8”
    - Select “RTSP” from drop-down on bottom left
    - Stream endpoint: rtsp://192.168.144.25:8554/main.264
    - Press Create
after a few seconds the live video thumbnail should be displayed (but it doesn't)

### Primary pain point(s)

Unable to view live video from an IP camera in cockpit

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2025-07-08:
can you confirm this @joaoantoniocardoso ?

joaoantoniocardoso, 2025-07-08:
The last change related to Redirect was to add the thumbnails (mavlink-camera-manager [`t3.19.2`](https://github.com/mavlink/mavlink-camera-manager/releases/tag/t3.19.2)), which arrived at BlueOS master from [this PR from May](https://github.com/bluerobotics/BlueOS/pull/3318), which corresponds to BlueOS [`1.5.0-beta.2`](https://github.com/bluerobotics/BlueOS/releases/tag/1.5.0-beta.2).

Another change that can relate is the change in the core version (from [this PR](https://github.com/bluerobotics/BlueOS/pull/3354) on June 5), which updates GStreamer version from `1.24.8` to `1.26.2`. This change arrived at BlueOS [`1.5.0-beta.6`](https://github.com/bluerobotics/BlueOS/releases/tag/1.5.0-beta.6).

My RTSP camera still work with Redirect on BlueOS `1.5.0-beta8`:

[image]

---

@rmackay9 could you: 
1. wait until the `Errors:` appear on the page and paste it here? Example:

[image]

2. test it with `1.5.0-beta.5` to see if that works?

joaoantoniocardoso, 2025-07-09:
That's intriguing... In this case, I'll need the mavlink-camera-manager logs from when it's failing on beta.8.

Yeah, the "RTSP Relay" is from an extension, I'll check how it's receiving the video. And yes, it's probably not deployed for arm64 (yet).

Thanks

rmackay9, 2025-07-10:
@joaoantoniocardoso,

And of course, now it starts working, at least partially
[image]

This is still an RPI5 running BlueOS-1.5.0-beta8 and the thumbnail is appearing now although I still can't make the video appear in cockpit.  The VideoPlayer is just a black square.  The only difference that I'm aware of is my precision landing extension which has been released now and is installed but not running.  Previously though it wasn't even installed.

For what it's worth, here's the log https://www.dropbox.com/scl/fi/vcpgqbio9clilkx92fh3i/mavlink-camera-manager.2025-07-10-11.log?rlkey=m1443rdd3kwivy1f063sw18v2&dl=0

joaoantoniocardoso, 2025-07-11:
Some observations from reading the logs:

* From 02:00 to 11:22, the MCM is struggling to have a stable connection to the camera via RTSP -- it is trying to connect to `rtsp://192.168.144.108", no port/path specified, which looks odd.

* After 11:22, it was, apparently, able to establish a connection, but it seems to be stuttering 5 times per second.
* At 11:24 you've opened the video page and some thumbnails/snapshots are being generated, which surprisingly makes the stuttering stop for a minute or so.
* At 11:25 the stuttering returns, and some WebRTC client starts talking
* At 11:26 there's some WebRTC negotiation, but it fails to establish aa connection (`WebRTCBin failed to negotiate under 10 seconds. Session will be killed immediatly to save resources`). This usually happens when (1) the client is denying all offered routes (192.168.144.20, 192.168.87.37, 192.168.42.1, 10.147.17.60 for both UDP and TCP). Cockpit has a configuration for that on the [Video Streams Configuration](https://blueos.cloud/cockpit/docs/latest/usage/advanced/#video-streams-configuration) page.

Given that, I'd ask you to check whether you have your vehicle's ethernet interface IP allowed on the Cockpit allowed WebRTC remote IP addresses list.

Thanks