bluerobotics/BlueOS #3448: bug: Raspi Kernel 6.12.34 breaks Ardupilot board detection

amarburg, 2025-07-29:
### Bug description

Was originally testing with a Raspberry Pi 5 with a vanilla "Bookworm 64-bit lite" image from Raspi Imager (rather than a pre-built BlueOS image), then manually installing per the [bootstrap instructions](https://github.com/bluerobotics/BlueOS/tree/master/install). 

The current version of this image installs the 6.12.34 kernel packages.

Ardupilot would not work on these installations with

```
Unknown board
```

in the ardupilot_manager log.

Per [ardupilot#30769](https://github.com/ArduPilot/ardupilot/issues/30769), Ardupilot's [`UtilRPI::_get_board_type_using_peripheral_base`](https://github.com/ArduPilot/ardupilot/blob/12ab5a9804838bdae53ef18185a19712ee346042/libraries/AP_HAL_Linux/Util_RPI.cpp#L39) function queries `/proc/device-tree/soc/ranges`

However, in the latest kernel image the DTB entry is renamed to `/proc/device-tree/soc@107c000000`, breaking board detection.

This bug does not affect the pre-built images from BlueOS, which contain the older 6.6.x kernels.  However if you `apt-get upgrade` it will install the newer image.

### Steps to reproduce

1. Create BlueOS SD card  using 1.4.2 image
2. Boot Pi5, running kernel 6.6.31
3. BlueOS successfully starts Ardupilot 4.5.3

[image]

5. On command line, `sudo apt-get update && sudo apt-get upgrade`
For me, this installs `linux-image-6.12.34+rpt-rpi-2712` and dependencies.

5. Reboot
6.  Confirm system is running 6.12.34
7. System cannot start Ardupilot:

[image]

### Primary pain point(s)

Nothing to prevent inadvertent upgrade to Pi OS, breaking system.

### Prerequisites

- [x] I have checked to make sure that a similar request has not already been filed or fixed.

patrickelectric, 2026-09-10:
This is related to trixie support, can you take a look @nukelet ?

joaoantoniocardoso, 2026-09-11:
~~@patrickelectric the bookworm situation (this issue) was probably solved recently.~~ nope, only for  `6.6.31`.

patrickelectric, 2026-09-11:
Ok, I removed the triage tag. 
Just to note, this issue is also important to help with Pi4 video encoding since the drivers are available in kernel 6.12.

nukelet, 2026-09-15:
Curiously, this doesn't seem to reproduce on a Raspberry Pi 4. I'm investigating further.

nukelet, 2026-09-16:
I ran some tests yesterday and it turns out that only the device tree for the Raspberry Pi 5 changed the name of the `soc` node when upgrading to kernel 6.12, so this really should only be an issue on the RPi5. It also seems that upstream Ardupilot already has a fix for this: https://github.com/ArduPilot/ardupilot/pull/30739, which has made it to a 4.6 stable release but not to 4.5, so we would need to backport it (should not be too difficult).